Noltari/pico-uart-bridge #10: Waits for Serial monitor?

ManuDLT, 2022-08-15:
I'm trying to program a CC1352 via pico-uart-bridge and using the SmartRF programmer (The official software for programming that chip) but nothing happens. 
I use an osciloscope to see the TX port output. When I click the "program" button, TX don't send anything, but when I open the serial monitor of Arduino, TX sends what is suppose to send when I clicked the "program" button.

Is there any instruction that makes the Pi Pico waits for the command of Serial Monitor open?

Noltari, 2022-11-04:
Could you try again with latest version?
There are a few improvements that could fix your issue.
Also, UART0 pins were changed, since I had some problems with the previous ones, which I couldn't reproduce with the new ones or with UART1.

Noltari, 2023-01-31:
Fixed by https://github.com/Noltari/pico-uart-bridge/pull/15

closed 2023-01-31 by Noltari