Noltari/pico-uart-bridge #9: Charters missing when using teraterm to send file using a physical loopback

lingshen1, 2022-08-15:
Test procedure:
1) Use a jumper header to shortcut pin 6 and pin 7 to physically loopback the TX/RX on UART1
2) Press and hold the key on the Raspberry PI PICO and plug in the USB cable. Send the "pico-uart-bridge-v3.0.uf2" file to the USB drive
3) Start Teraterm, "New Connection" and select the port with the loopback and set baudrate to be 115200 BPS and no flow control
Make sure when you type, you can see echo in the window, which indicates the physical loopback is working. If not, choose the other USB uart port
4) In the Teraterm, "File->Send File..."->Select the file "wait_regex.ttl" from the default path (from Teraterm installation place). You will see character missing as listed below.
5) I tested using a CP2102 USB UART using same loopback test procedure and did not see any character missing

# Original File sent by Teraterm (no line/character delay)  
; sample macro of Tera Term
;
; File: wait_regex.ttl
; Description: wait command with regular expression
; Environment: generic
; Update: 2005/10/15
; Author: Yutaka Hirata

connect '192.168.1.2 /2 /ssh /auth=password /user=yutaka /passwd=yutaka'

while 1

	sendln 'uptime'
;	waitregex 'Longhorn' '.*@sai' 'Pentium\d+' 'TeraTermX{3}'
	waitregex '(You).*(log\w+).*(this)'

	int2str s result
	messagebox s 'result'
	messagebox inputstr 'inputstr'
	messagebox matchstr 'matchstr'
	messagebox groupmatchstr1 'groupmatchstr1'
	messagebox groupmatchstr2 'groupmatchstr2'
	messagebox groupmatchstr3 'groupmatchstr3'
	messagebox groupmatchstr4 'groupmatchstr4'

	sendln 'cal'
	wait 'yutaka'

	pause 3

endwhile

# Actual shown in the window and you will see quite a lot of characters are missing. If in the "Setup->Serial" and put 1ms delay on each line, then no character missing happens. It might be (my ) some issues in the FIFO management or some code is blocking the execution for too long and causes the FIFO overflow. 

; sample macro of Tera Term
;
; File: wait_regex.ttl
; Descripion: wait command with regular e
; Update: 2005/10/15
; Author: Yutaka Hirata

connect '192.168.r=yutaka /passwd=yutaka'

while 1

        sendln 'uptime'
;       waitregex' 'TeraTermX{3}'
        waitregex '(Yo).*(log\w+).*(this)'

        int2str smessagebox inputstr 'inputstr'
        messagebox matchstr 'matchstr'
        matchstr1'
        messagebox groupmatctr2 'groupmatchstr2'
        messagebox
        messagebox groupmatchstr4 'groupmatchstr4'

        sendln 'cal'
        wait 'yutaka'

        pause 3

endwhile

JoeSc, 2022-09-22:
@lingshen1 I encountered the same data loss as you described.  In searching I found another USB-UART repository that appears to work better https://github.com/harrywalsh/pico-hw_and_pio-uart-gridge  Then I forked that repository since I saw similar but not as bad data loss.  

My fork is https://github.com/JoeSc/pico-sexa-uart-bridge I have confirmed 6x receivers at 115200 baud results in no packet loss with ~400KB file transferred.  And similarly 6x loopbacks works.  Do you want to give it a try?

lingshen1, 2022-10-01:
I will give it a try. Thanks a lot.

lingshen1, 2022-10-20:
Your branch works great. I tested all 6 channels using the same way I
tested it on the Noltari's branch by sending files in full speedin Teraterm
and it worked as expected.
Thanks a lot!
Ling

Noltari, 2022-11-04:
Thanks for your code @JoeSc,
I've made some improvements based on your work, but I won't be adding PIO support since 2 UARTs is more than enough for me and users can always use yours for more UARTs.
Anyway, I'm closing this since the missing characters issues should be resolved in the latest version.

closed 2022-11-04 by Noltari