Noltari/pico-uart-bridge #11: Does not recognize in Windows after unplugging and replugging.

dontdeadopeninside, 2022-10-08:
Title. Need to hold down BOOTSEL to get it to recognize as a USB device. Then requires reflashing. After flashing, unplugging and replugging doesn't even make the Windows USB insert noise, and device manager shows no changes.

Noltari, 2022-11-04:
You need to hold BOOTSEL down to flash it, as described in https://learn.adafruit.com/adafruit-hallowing/uf2-bootloader-details

randy408, 2023-06-25:
I'm having the same issue, the COM ports stop showing up in Device Manager after replugging. Very rarely the LED lights up (what does it mean?) after copying the .uf2 in BOOTSEL mode but couldn't confirm if the serial bridge is working.

After a couple tries of reflashing and replugging the LED didn't light up anymore. I checked if the Pico's are still functioning by flashing a blinking LED sample and they still work after multiple power cycles.

ma261065, 2023-12-14:
Not sure why this was closed - it is still an issue. Since v2.1 resetting the RPi2040 somehow loses the code that was flashed to it.

Repro steps:
1) Press and hold BOOT and press and release RESET
2) D: shows up in Windows
3) Drop pico-uart-bridge-v4.1.uf2 on to D:
4) The file copies and device restarts. COM18 & COM19 show up in Device Manager and can be used just fine
5) Either press RESET or unplug/replug the RpI2040 - COM18 & COM19 disappear and do not re-appear in Device Manager

The only way to use it again is to re-do steps 1 - 4

Noltari, 2023-12-14:
> Not sure why this was closed - it is still an issue.

Because I couldn't reproduce it on my device.

ma261065, 2023-12-14:
How can I help you to troubleshoot this? The device I'm using is this one - https://www.waveshare.com/rp2040-zero.htm

criptych, 2024-04-10:
> As mentioned, v2.1 does not show the same behaviour, but all subsequent releases do.

I had the same problem with an Adafruit KB2040; this observation got me to a solution (or at least a workaround).  One of the most notable differences between v2.1 and v3.0 is the use of `set_sys_clock_khz` to change the clock speed to 250 MHz.  Commenting/removing this line allowed Windows to still recognize the board after disconnecting and reconnecting it.  

I'm not sure why the extra speed is needed -- perhaps to run both UARTs at high speeds? I'm only using one and only up to 921600 baud, and haven't noticed any issues.

closed 2024-04-11 by Noltari