Noltari/pico-uart-bridge #12: Flashing ESP Device through ESPtool

mikeneiderhauser, 2022-12-24:
Looking for some assistance using your pico-uart-bridge to flash an ESP device via esptool.

I currently keep getting timeouts through ESPtool.  When I connect to minicom and look at the ttyACM0 device and power cycle the ESP chip on UART0, i do see activity, but can't seem to get flashing via ESPtool to work.

Any advice on getting this to work with esptool?

Thanks!

Noltari, 2023-01-31:
@mikeneiderhauser after merging https://github.com/Noltari/pico-uart-bridge/pull/15 I could flash an ESP-01s through the ESPHome website (although it finished with an error it flashed succesfully), but I couldn't flash it with the ESPHome flasher.

Also sometimes I can see some corruption when connecting through serial port:
[image]

mikeneiderhauser, 2023-02-01:
I updated to the latest release u2f and attempted to flash again.  I still can't seem to get it to connect.

tzhvh, 2023-03-16:
It's a bit flaky but I tried  `esptool.py --port /dev/ttyACM0 --baud 76800 --chip esp8266 --before no_reset_no_sync --connect-attempts 0 read_mac` and it worked okay for my use-case, perhaps flashing would work as well

wiredopposite, 2024-02-05:
I'm having this issue with esptool as well, Adafruit web serial esptool does seem to work though. I haven't seen it fail or do anything weird.

therealdreg, 2025-08-31:
for esp check my fork: https://github.com/therealdreg/pico-esp-flasher