Noltari/pico-uart-bridge #14: Trying to quickly send more than 3 characters results in corruption

lmore377, 2023-01-22:
This might explain #10 #12 and #13. Basically if you try to send more than 3 characters at once the datastream gets corrupted and characters either start sending out of order or just get entirely replaced. This happens on linux (not sure about windows) and you can see these results by doing loopback on either serial port, opening it in a serial terminal emulator like gtkterm and pasting a more than 3 characters or using echo like below. This happens with any baud rate

Here's some examples:

`echo 123 > /dev/ttyACM0` = No corruption
```
123

```
`echo 1234 > /dev/ttyACM0` = 3&4 get dropped and there's a random newline
```
12

```

`echo 12345 > /dev/ttyACM0` = Everything except 1&2 is replaced with newlines
`echo "this is an example of text" > /dev/ttyACM0` = Everything except th is replaced with newlines

mgduda, 2023-01-29:
I'm seeing this same behavior as well. Receiving characters through the UART seems to have no issue, but sending bursts of more than a few characters out of the UART results in the behavior noted by @lmore377.

I did try enabling the FIFOs on the UARTS (by modifying the [call to `uart_set_fifo_enabled`](https://github.com/Noltari/pico-uart-bridge/blob/v4.0/uart-bridge.c#L292) in `init_uart_data`), and that allowed 32 - 33 characters to be transmitted correctly; but further characters in the message were then sent as the last character in the message.

Noltari, 2023-01-31:
Fixed by https://github.com/Noltari/pico-uart-bridge/pull/15

closed 2023-01-31 by Noltari