Noltari/pico-uart-bridge #16: flashing Beken BK7231N device

rew1ndustry, 2023-04-15:
hello, this is parallel to the ESP question still open.

am trying to use pico-uart-bridge to flash beken chips, and so far no luck.

unfortunately i do not own any other uart-usb bridges, so am unable even to determine if the pico is the problem.

the BK7231N comes with two UARTs, one is the flash port and the second is used for debugging.

am able to read the debugging port on the second channel, so am fairly sure the pico bridge is working.

am now working through your code, looking for anything that might help me to work out what the problem is, and already i have a question - please forgive me, i am very new to this hardware level coding..

at line 304 in uart-bridge.c, you call `set_sys_clock_khz(250000, false);`

however i believe the picoW has a max clock rate of 133MHz

am i just not understanding how this works?

Goldenkrew3000, 2024-02-20:
Hey. Idk if you are still looking for a solution, but I was having the exact same issue. While I didn't get it working with the pico, the knock-off FTDI FT232 Usb boards work just fine for this :)

OoSTARTERoO, 2024-09-15:
I also had the problem, but somehow ltchiptool CLI version worked
[ltchiptool docs](https://docs.libretiny.eu/docs/flashing/tools/ltchiptool/#cli-usage)