Noltari/pico-uart-bridge #17: Open Collector TX

muebau, 2023-05-21:
What would be needed to (if possible) to change the TX to an open collector like output.

I TX is LOW the output pin would be GND and if TX would be HIGH the pin would be floating (eg. high impedance, Z) but not VCC.

I thought one way to implement this could be a simple 'copy' PIO code which simply mirrors the TX state but gets 'Z' instead of HIGH (VCC).

Is there a simple way to implement this?

laf0rge, 2023-11-29:
AFAICT the rp2040 doesn't do open-collector/open-drain.  So you have to emulate it by switching the TX pin to an input whenever you're not transmitting.  That would have to be done at a very low level (inside the driver) as it has to reflect the actual UART hardware IP cores "transmit idle" state, and not just some higher-level state at which point the application code has told the UART driver to write something.

In the end, this might be easier by using a modified PIO soft-uart instead of the hardware UART IP core.

xxxajk, 2024-04-15:
...or just add a diode and a pullup resistor of about 1k, which emulates an open collector.