Noltari/pico-uart-bridge #20: Communication stuck if connecting to UART1 only

alessandrodn, 2024-02-04:
First of all, kudos for the project.

I have a device exposing two UARTS and I connect both the serials to the PICO.
I noticed that if I open a serial session (I use `screen` on macOS) to UART 1 only, the communication sticks after few bytes exchanged.
If there is an open session to UART0 everything is fine.
It's also fine to open a session to UART0 only.

xxxajk, 2024-03-01:
Did you try updating the pico SDK to HEAD? It includes lots of fixes, probably might fix it for you.
It's one of the first things I do when I run into problems.

xxxajk, 2024-03-01:
I was also able to recreate the same problem here on Linux, but there's specific conditions to trip the problem.

The communication packets have actually started, then stopped from the PC side...
...while there is a data flood on either or both channels from the UARTs.

This is caused by two bugs.

Basically the code spins waiting to unblock, which will never happen until the host catches up, it never will. Basically it gets stuck because...
...the silly read ISRs that end up in an endless loop. 
This is  because the data incoming from the UART is not discarded if the buffer is full, thus, the IRQ is never cleared.

The fix is to remove the infinite loop by dropping the UART data on the floor.

xxxajk, 2024-03-01:
Try this version and let me know if it works.
My changes are as follows:
```
UART0_TX 0
UART0_RX 1
UART0_LED_TX 19
UART0_LED_RX 18
   
UART1_TX 4
UART1_RX 5
UART1_LED_TX 17
UART1_LED_RX 16
```
It is also possible to send breaks with this firmware.

[uart_bridge.uf2.zip](https://github.com/Noltari/pico-uart-bridge/files/14460609/uart_bridge.uf2.zip)

Noltari, 2024-04-11:
> Try this version and let me know if it works. My changes are as follows:
>
> ```
> UART0_TX 0
> UART0_RX 1
> UART0_LED_TX 19
> UART0_LED_RX 18
>
> UART1_TX 4
> UART1_RX 5
> UART1_LED_TX 17
> UART1_LED_RX 16
> ```
>
> It [...]

@xxxajk it would help a lot if you could provide the modified source code with the fixes.

xxxajk, 2024-04-14:
Done.
https://github.com/xxxajk/pico-uart-bridge
If it is a fix for enough people, I'll do a pull request to here, although there's not a lot of activity here...

Noltari, 2024-04-15:
> Done.
> 
> https://github.com/xxxajk/pico-uart-bridge

Thanks!

> If it is a fix for enough people, I'll do a pull request to here, although there's not a lot of activity here...

Yes, please, open a new PR with this

xxxajk, 2024-04-15:
Are you OK with the different pin setup?

alessandrodn, 2024-09-27:
I can confirm that #25 fixes the issue 🎊

xxxajk, 2024-09-28:
Glad someone cherry picked it. 
Couldn't stand the formatting at all, which doesn't follow best practices. 
That's all good though, and why we got forks ;-)

Noltari, 2026-11-08:
Should be fixed by dd641210a8c6cc2fa92c4b25e585fd1ecbcc1371

closed 2026-11-08 by Noltari