Noltari/pico-uart-bridge #21: BUFFER SIZE=2560 line 20 in uart-bridge.c ?

patfrench, 2024-03-08:
Hello,

I'm trying to understand the source code and while reading I saw this :
https://github.com/Noltari/pico-uart-bridge/blob/ca81e5c242148a4e2462a10360f886cc8849a1fa/uart-bridge.c#L20

Why 2560 ?

Your project is awsome

Noltari, 2024-03-10:
> Hello,
> 
> I'm trying to understand the source code and while reading I saw this :
> 
> https://github.com/Noltari/pico-uart-bridge/blob/ca81e5c242148a4e2462a10360f886cc8849a1fa/uart-bridge.c#L20
> 
> Why 2560 ?
> 
> Your project is awsome

Someone suggested increasing it some time ago and I just did it :)

patfrench, 2024-03-10:
Hello
I'm sorry if this is a stupid question, but I thought we were limited to the UART buffer.

Thank you

Noltari, 2024-03-10:
> Hello I'm sorry if this is a stupid question, but I thought we were limited to the UART buffer.
> 
> I just saw that the buffer is 256 so I guess a typo.
> 
> Thanks again

Yes, but you might not be able to forward that data to the USB host, so you might need to store more than the UART buffer.

closed 2024-03-10 by Noltari