Noltari/pico-uart-bridge #22: USB-RS232 console port?

aidzer0, 2024-03-25:
Hello,

This is not an issue so much so as a question. 

Could the pico be used as a RS232 FTDI converter for console connections to network gear? 

As an example the pico would replace the need for this type of cable:
https://www.startech.com/en-ca/cards-adapters/icusb232v2

I see that pin 2/3 on the DP9 cable are RX/TX so could I just connect those to pins 16/17 or 6/7 on the Pico and connect using putty? the cable would not include a serial/FTDI chip though. it would be a usb mini to DP9 female that would need a serial to usb converter (I am hoping the pico would be that part).

Thank you!

muebau, 2024-03-29:
I think the adapter you mention consists of the actual USB to serial bridge (TTL level) and a level shifter.

Common USB to serial bridges would be chips like CP21xx, PL2303, FTDI*, CH341, etc. which give a 5V or 3.3V TTL level. They get combined with a level shifter like the popular MAX232 which gives the actual RS232 levels.

So if you would combine the RPi Pico with a MAX232 it should be equivalent to such adapters indeed.

aidzer0, 2024-03-29:
Hello, 

Yes you are correct, So I would need the MAX232 to do what I need. Thank you for the response!

I will close the issue now. Have a great weekend!

closed 2024-03-29 by aidzer0