Noltari/pico-uart-bridge #23: Maximum baud rate

anonymix007, 2024-04-15:
What will be the maximum achievable baud rate?  I've seen reports of ~1 MBaud, but for my application 3 or more will be needed. Can it be achieved on RP2040?

xxxajk, 2024-04-19:
Providing that you have proper wiring, yeah... But you really want to wait until bug #20 is fixed with https://github.com/Noltari/pico-uart-bridge/pull/24 or just use my branch until that point.

csarn, 2025-04-14:
I tried it with a short loopback wiring (connecting both pico uarts together), and had increasingly severe transmission errors starting at 1MBaud. Then I pulled the changes from the mentioned #24 (or rather #25), and had zero transmission errors (I tried up to 5Mbaud).

xxxajk, 2025-04-15:
Yeah, remembering to have the pullup enabled is the trick to that working in my original patch.